StephanJoubert/home_assistant_solarman #660: Certain spikes in some values

MikealSouth, 2024-09-17:
Hi, I am using Solarman 1.0.0 on a up2date HA installation and I am experiencing spikes in the daily_load_consumption and daily_energy_sold values, please see the attached screenshot.
[image]
Could someone please point to me what I am doing wrong?
I've started using the Solarman Integration around a year ago, a set-it-and-forget-it process.
This started around 4-6 weeks ago.

PV-Joe, 2024-09-24:
Try this more advanced and heavily maintained version. It runs very stable with SG04LP3:

https://github.com/davidrapan/ha-solarman

MikealSouth, 2024-09-30:
Interesting, so unless I send more to the grid then my PV production plus my battery capacity is this version also has the same issue.
As they are reading the data from the inverter I am sure that it's the inverters problem.
As it's still under warranty I would ask Deye but I am sure that this way of data collection this isn't supported.

Is there a official tool where I can see this data