StephanJoubert/home_assistant_solarman #664: wifi plug pro-05 dongle, web portal

PieterBeukesAI, 2024-09-25:
I have a plug pro-05 dongle connected to my inverter. It is currently setup via cloud to display data on the SMART ESS APP. Im trying to set it up on solarman but cannot access the device web portal. 

The device IP is currently 10.0.0.29

I get the following errors:

URL 10.0.0.29
[image]

URL 10.0.0.29:8899
[image]

Ive tried to setup the device in Solarman with its current IP but no data is pulling through
[image]

[image]

Any suggestions will be appriciated.
Thanks

innomaxx, 2024-10-17:
Hi @PieterBeukesAI, any progress? We also have wifi plug pro-05 dongle and would like use it with HA

PieterBeukesAI, 2024-10-18:
@innomaxx  Unfortunately not, still have the same issue.

brunoclc, 2025-02-09:
For me until there, https://github.com/SilverFire/dessmonitor-homeassistant/tree/main?tab=readme-ov-file is the only option