StephanJoubert/home_assistant_solarman #669: Didnt work anymore with HA 2024.9.3 and Solarman 1.5.1

kahnbeinKai, 2024-10-04:
Hello,
after an update of HA the integration wont work again, all sensors have unknown values, befor the update everything works fine.

Here ist my Log file from HA:
```
2024-10-04 20:07:49.559 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-04 20:07:51.498 ERROR (MainThread) [homeassistant.components.sensor] The systemmonitor platform for the sensor integration does not support platform setup. Please remove it from your config.
2024-10-04 20:07:51.565 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/deye_string.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 34: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 34: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
```

gardiol, 2024-10-07:
Nothing in the log seems unusual or wrong.
Something else must be your issue i think

kahnbeinKai, 2024-10-08:
Ok, thank you for your reply. 
This is my full log, maybe you find a issue here ? Like i said, befor the update of HomeAssistant everythink works perfect. 
`2024-10-04 20:19:16.103 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-04 20:19:18.069 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/deye_string.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 34: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 34: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 308, in async_setup_platform
    await self._platforms[key].async_setup(platform_config, discovery_info)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 298, in async_setup
    await self._async_setup_platform(async_create_setup_awaitable)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 358, in _async_setup_platform
    awaitable = create_eager_task(awaitable, loop=hass.loop)
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/config/custom_components/solarman/sensor.py", line 84, in async_setup_platform
    _do_setup_platform(hass, config, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
  File "/config/custom_components/solarman/solarman.py", line 34, in __init__
    with open(self.path + self.lookup_file) as f:

2024-10-04 20:19:21.817 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ states('sensor.tasmota_sm_1_8_0') | float / 1000 }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2273, in forgiving_float_filter
    return float(value)
           ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 619, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2654, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 939, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2276, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1887, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states('sensor.tasmota_sm_1_8_0') | float / 1000 }}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 731, in async_render_to_info
    render_info._result = self.async_render(  # noqa: SLF001
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 238 lines cut ...]
2024-10-07 12:17:36.296 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.bosswerk2_pv1_voltage_2 is taking over 10 seconds
[... cut ...]
2024-10-07 18:38:13.168 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.bosswerk2_pv1_voltage_2 is taking over 10 seconds
2024-10-07 18:39:43.183 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.bosswerk2_pv1_voltage_2 is taking over 10 seconds
2024-10-07 18:39:57.018 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-10-07 18:49:19.369 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.deye1_pv1_voltage is taking over 10 seconds
`

gardiol, 2024-10-08:
I don't know, i see quite many errore on this log, not even clear if they all refer to the same extension or not, i would check them one one by one and see if they are fixable or not. I don't see anything specific for home_assistant_solarman except a generic error that says unable to read data within xxx seconds. But i can't help more, sorry! Maybe maintainer will pop up.

kahnbeinKai, 2024-10-08:
Thank you for your help.
I did a Core upgrade to 2024.10.1 and now they work all again .... 
The error log looks the same, most of the errors are custom sensors not related to solarman. 
I will not do a upgrade of my home assistant again :) 
Best regards
Kai