StephanJoubert/home_assistant_solarman #671: INVT XD6KTL solar inverter

MrLopHA, 2024-10-06:
Hello everybody

I have an INVT inverter installed and I see the production data via the SOLARMAN app. 

I need to integrate the inverter into Home Assistant (running on RasbPi4) so that my car charger receives data from the inverter. 

The integration via HACS does not work for me. Has anyone got this inverter and managed to integrate it in HA? Or extract your data via Modbus? 

I have also written to SOLARMAN to ask them about their API, but so far I have no reply. 

Thank you very much!

PV-Joe, 2024-10-06:
This type of inverter is not confirmed as compatible yet in the ha-Solarman integration.

https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices

Maybe you want to try after installation if any of the profiles work for you?

Install from here:
👉 https://github.com/davidrapan/ha-solarman

MrLopHA, 2024-10-06:
Hi Joe!

Thank you very much for your recommendation. I have installed the integration and when I run it, it asks me for the inverter port (default is 8899) but I don't know if it is the correct port and I can't find it anywhere.

Can you help me with that? the integration is not working at the moment, but it could be because of my misconfiguration of the data it asks for.

[image]

PV-Joe, 2024-10-06:
Hi again,

try 8899 it should be working.

For further support, go here:
https://github.com/davidrapan/ha-solarman

Maybe David has an idea to connect properly to your Solarman Stick-Logger.

MrLopHA, 2024-11-11:
I come back here to say that the problem has been solved thanks to the integration of David Rapan.

The thread can be closed.

thaibacgiang, 2024-11-24:
I am using Imars MG5KTL inverter with monitoring on solarman, please help me put this device on home assistant, thank you