StephanJoubert/home_assistant_solarman #674: Detected blocking call

stamatakisa, 2024-10-06:
Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/deye_hybrid.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2253, in _async_forward_entry_setup await entry.async_setup(self.hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 131, in async_setup_entry return await hass.data[DATA_COMPONENT].async_setup_entry(entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 194, in async_setup_entry return await self._platforms[key].async_setup_entry(config_entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 328, in async_setup_entry return await self._async_setup_platform(async_create_setup_awaitable) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 358, in _async_setup_platform awaitable = create_eager_task(awaitable, loop=hass.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/solarman/sensor.py", line 89, in async_setup_entry _do_setup_platform(hass, entry.options, async_add_entities) File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file) File "/config/custom_components/solarman/solarman.py", line 31, in __init__ with open(self.path + self.lookup_file) as f:

svenvdc, 2024-10-07:
Same issue,

Logger: homeassistant.util.loop
Source: util/loop.py:136
First occurred: 12:58:40 (1 occurrences)
Last logged: 12:58:40

Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/solis_hybrid.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2160, in async_forward_entry_setups await self._async_forward_entry_setups_locked(entry, platforms) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2171, in _async_forward_entry_setups_locked await asyncio.gather( File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2173, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2253, in _async_forward_entry_setup await entry.async_setup(self.hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 131, in async_setup_entry return await hass.data[DATA_COMPONENT].async_setup_entry(entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 194, in async_setup_entry return await self._platforms[key].async_setup_entry(config_entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 328, in async_setup_entry return await self._async_setup_platform(async_create_setup_awaitable) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 358, in _async_setup_platform awaitable = create_eager_task(awaitable, loop=hass.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/solarman/sensor.py", line 89, in async_setup_entry _do_setup_platform(hass, entry.options, async_add_entities) File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform inverter = Inverter(path, inverter_sn, inverter_

ababak, 2024-12-13:
As far as I can tell, it’s a cosmetic issue since it occurs within the constructor once at startup and not in the main loop.