StephanJoubert/home_assistant_solarman #675: untimely disconnection

William95270, 2024-10-06:
Hello everyone, 

It seems to work well on my side, I have all the values ​​going back but I have a lot of disconnections throughout the day. Sometimes for a minute, sometimes it lasts up to 20 minutes. This can disrupt automations that I would like to set up. Do you know why I have all these disconnections?

[image]

gardiol, 2024-10-07:
This extension seems quite abandoned. In my experience, it means you are requesting info too fast to the Solarman stick. Lowering the request time (see https://github.com/StephanJoubert/home_assistant_solarman/issues/238) could improve.

William95270, 2024-10-07:
@gardiol thx. 
Actually, I made a mistake, I’m not using this service, but this one: https://github.com/davidrapan/ha-solarman. I’m not sure if the information you provided is also valid for this one

davidrapan, 2024-11-16:
Hi @William95270, yes it is. 😉