StephanJoubert/home_assistant_solarman #676: SOLARMAN

tomsi62, 2024-10-12:
Hello,
I have installed the solarman app but on homeassistant no data are shown. Can you please help me?[image]

davidrapan, 2024-11-27:
Hi @tomsi62, try [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) be careful with selecting proper profile for your device. 😉

tomsi62, 2024-11-27:
Thank you Stephan 🙏. What profile shall I take for my 10kW inverter?

davidrapan, 2024-11-27:
I'm not Stephan. 😉 It depends on brand and model of your inverter. Rated power is in this case irrelevant.

tomsi62, 2024-11-28:
Sorry, I mean David 😉

oHo2go, 2024-12-13:
Hi, I am having the same problem, but as I understood it from the instructions, the device serial number should be used, but mine is not in the range that is suggested in the instructions. There is an alternative path hinted to, but I could not find any details. I am new to this, so the problem is certainly infront of the system...