StephanJoubert/home_assistant_solarman #677: Integration was not loaded anymore after update to homeassistant:latest

DBREngineer, 2024-10-13:
I got these messages:

```
Logger: homeassistant.setup
Source: setup.py:269
First occurred: 19:23:26 (2 occurrences)
Last logged: 19:23:29

Setup failed for custom integration 'solarman': Requirements for solarman not found: ['pysolarmanv5'].

```

Has anyone seen the same. I went back to the last docker image I used (Septemeber 2024, Core 2024.09.2 / Frontend 20240909.1) and everything workes fine again.

Coolen07, 2024-10-16:
i have the same problems.. you have to downgrade to 24.10.10 that works for me...the newer versions are not running anyomer

ayubn, 2024-10-29:
i hv same issue no data after upgrading

Shelis1976, 2024-11-09:
I have new instalation so I have ver 2024.11  and I cannot downgrade.... any advice? solarman not working

ayubn, 2024-11-18:
> I have new instalation so I have ver 2024.11 and I cannot downgrade.... any advice? solarman not working

use this command in home assistant terminal and restart home assistant 

ha core rebuild

PiotrMilosz, 2024-12-04:
> > I have new instalation so I have ver 2024.11 and I cannot downgrade.... any advice? solarman not working
> 
> use this command in home assistant terminal and restart home assistant
> 
> ha core rebuild

Does it help? I have had the same issue and I rolled back. Do You know if the problem exists on the newest build of HA?

ayubn, 2024-12-04:
> Does it help? I have had the same issue and I rolled back. Do You know if the problem exists on the newest build of HA?
yes its workingggg for me
ha core rebuild