StephanJoubert/home_assistant_solarman #685: Incorrect values of Load L1, L2, L3 Power sensors for deye_sg04lp3.yaml

SmolikOne, 2024-11-28:
Hello. I have Deye SUN-12K-SG04LP3-EU/AU (no battery yet connected) integrated with your power flow card using deye_sg04lp3.yaml, but I have one issue with collected data. Namely when the solar production is delivering power above several hundred Watts per phase, values of generated power on Load L1, L2, L3 Power sensors and in Total Load Power sensor shows crazy data. So for example if PV load on phase 1, 2, 3 is lower than ca. 200W per phase, Load L1, L2, L3 Power sensors are showing correct data of power usage per particular phase. But when PV Load on phase 1, 2, 3 is higher than ca. 200W per phase, then Load sensors are showing 65.46kW load per phase which is of course incorrect. The same applies to Total Load Power sensor, that is showing combined over 194kW incorrect load. What is intersting Load data per phase on the inverter itself is showing correct data, so the question is why Solarman Integration is showing such absurd values on L1, L2, L3 Load? It there some place where I can correct it myself, or it is a error on Solarman Integration side? Thank you in advance.

[image]
[image]

davidrapan, 2024-11-28:
Hi @SmolikOne, if you want you can try switch to [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman). I have the same device.

SmolikOne, 2024-11-28:
Thx davidrapanI've, I've installed it today and I need to wait for tommorows PV load to check if the values will correct or not.

rene-neimann, 2024-12-09:
After upgrading to 2024.12, I have very similar issue using deye_hybrid.yaml with my Deye SUN-12K-SG04LP3-EU.

davidrapan, 2024-12-10:
> After upgrading to 2024.12

Upgrading of what?

rene-neimann, 2024-12-10:
> > After upgrading to 2024.12
> 
> Upgrading of what?

Sorry, I meant upgrading to Core 2024.12.0 (I run HA Core docker).

davidrapan, 2024-12-10:
Version of HA or update of HA has nothing to do w/ how sensors work and which value they display.

rene-neimann, 2024-12-10:
> Version of HA or update of HA has nothing to do w/ how sensors work and which value they display.

I'm just trying to describe the issues that seem to be related to HA Core 2024.12.0 upgrade and possibly the underlying Python version upgrade. I didn't make any other changes in the system that could affect this. I also have another HACS integration which broke after 2024.12 upgrade and it's been determined to be because of numpy version upgrade: https://github.com/home-assistant/core/issues/132347

As a sidenote - I'm also the one who reported a possible bug in your Solarman integration and I'm about to test your fix: https://github.com/davidrapan/ha-solarman/issues/312

SmolikOne, 2024-12-10:
For me, after switching to https://github.com/davidrapan/ha-Solarman version, everything is working correctly. I've also updated to HA 2024.12 and it didn't break anything.