StephanJoubert/home_assistant_solarman #690: Have message like Querying [250 - 279] failed from time to time in logs

sirmax123, 2024-12-04:
From time to time I see messages like shown below:

```
2024-12-04 11:20:14.163 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.deye_solarman_pv1_power is taking over 10 seconds
2024-12-04 11:20:20.063 WARNING (SyncWorker_60) [custom_components.solarman.solarman] Querying [250 - 279] failed with exception [Empty: ]
2024-12-04 11:20:20.063 INFO (SyncWorker_60) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.31.245:8899
2024-12-04 11:20:20.064 WARNING (SyncWorker_60) [custom_components.solarman.solarman] Querying [250 - 279] failed, [1] retry attempts left
2024-12-04 11:20:20.064 INFO (SyncWorker_60) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.31.245:8899
```
It is ~81 fail
```
docker logs ea783a24a1c7 2>&1 | grep 'custom_components.solarman.solarman' | grep fail | grep '2024-12-04' | wc -l
81
```

Could you please provide me any suggestion where to start an investigation?

I checked network connectivity and did not find any packet loss with ping big packets
```
ping 192.168.31.245 -s 1472 -i 0.1 -n
```

Is it expected when the logger stops responding (data reading, not ping)  from time to time? Or should I replace the logger?

davidrapan, 2024-12-04:
Hi, @sirmax123, what profile it is?

sirmax123, 2024-12-04:
Hi, @davidrapan!

It looks like it is `deye_hybrid.yaml`
[image]

The data looks correct for the first view.

I created my own small script to test and also got exceptions that look like I see in logs

Here is part of code, and it also lost data from time to time with 
``
## 2024-12-04 10:29:08,704 collect_data_therad [read_registers] pysolarmanv5.pysolarmanv5.NoSocketAvailableError: Sleeping for 75 seconds. Registers from 0 to 124, attempt: 9
```
```
     for read_number in range(1, self.inverter_registers_reads_number+1):
            read_attempts = self.max_read_attempts
            while read_attempts:
                try:
                    if not modbus:
                        modbus = pysolarmanv5.PySolarmanV5(
                            self.stick_logger_ip, self.stick_logger_serial,
                            port=self.port, mb_slave_id=self.mb_slave_id,
                            verbose=self.verbose, logger=self.logger,
                            auto_reconnect=self.auto_reconnect,
                            v5_error_correction=self.v5_error_correction
                        )
                    self.logger.info("[read_registers] Read attempt: {}, Read number: {}. Rading registers from {} to {}".format(
                            read_attempts,
                            read_number,
                            read_start_register,
                            read_start_register + self.max_number_of_registers_to_read_in_request -  1
                        )
                    )
                    inverter_read_raw_result = modbus.read_holding_registers(
                        register_addr=read_start_register,
                        quantity=self.max_number_of_registers_to_read_in_request
                    )
                    self.logger.debug("[read_registers] Read result (raw): {}".format(inverter_read_raw_result))
                    self.inverter_read_raw_result_all_registers = self.inverter_read_raw_result_all_registers + inverter_read_raw_result
                    read_start_register = read_start_register + self.max_number_of_registers_to_read_in_request
                    self.logger.info("[read_registers] Finished reading registers from {} to {}".format(
                            read_start_register,
                            read_start_register + self.max_number_of_registers_to_read_in_request - 1
                        )
                    )

                    break
                except pysolarmanv5.pysolarmanv5.NoSocketAvailableError as E:
                    random_sleep = random.randint(0, self.sleep_on_inverter_read_error)
                    sleep = self.sleep_on_inverter_read_error + random_sleep
                    self.logger.error("[read_registers] pysolarmanv5.pysolarmanv5.NoSocketAvailableError: Sleeping for {} seconds. Registers from {} to {}, attempt: {}\n {}".format(
                            sleep,
                            read_start_register,
                            read_start_register + self.max_number_of_registers_to_read_in_request - 1,
                            read_attempts,
                            E
                        )
                        ....
```

So, can it be something wrong with the wifi stick?
 Is there any reason to use an ethernet (wires, not wireless) stick instead? I can purchase one but not sure is it supported.
As far as I understand, it is possible to use pure Modbus instead of solarmanv5?

Or may be I'm doing read in the completely wrong way?

davidrapan, 2024-12-04:
Hmmm, could you try to run [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) instead, and see if you'll get the same result? (If you decide to do se leave profile selection to auto)