StephanJoubert/home_assistant_solarman #691: Integration Solarman not Working

Freitag29, 2024-12-12:
Hi everyone,
I have installed the Solarman integration and configured the device, but all sensors are in an unknown state.
I tried modifying the YAML configuration file with different versions, but it didn’t work. I also attempted to configure the device using auto-discovery, but nothing changed.

Here is my system information:

Core: 2024.12.2
Supervisor: 2024.11.4
Operating System: 14.0
Frontend: 20241127.7

Can anyone help me?

PV-Joe, 2024-12-12:
Check out this integration:

https://github.com/davidrapan/ha-solarman

It works well and is updated regularly.

Freitag29, 2024-12-13:
Thanks for the suggestion.
I have installed the integration but it not work .
[image]

Ros-83, 2024-12-14:
Anche a me da ieri, dopo l'ultimo aggiornamento del core a 2024.12.3 ha smesso di funzionare. C'è modo di riparare?

FLconsultant, 2024-12-18:
> Anche a me da ieri, dopo l'ultimo aggiornamento del core a 2024.12.3 ha smesso di funzionare. C'è modo di riparare?

Anche a me stesso problema. 
Ho risolto con l'aggiornamento del core a 2024.12.4

Freitag29, 2024-12-18:
Ciao alla fine a furia di smanettare tra le configurazioni dell'integrazione sono riuscita a farla partire.

kappa1315, 2024-12-29:
Ho appena istallato Solarman, ma ho tutti i sensori in unknow, come posso fare per visualizzarli, integrazione solarman e home Assistant aggiornati all’ ultima versione

catalindolh, 2025-03-06:
> Ciao alla fine a furia di smanettare tra le configurazioni dell'integrazione sono riuscita a farla partire.

Can you tell me more exactly how it was resolved?