StephanJoubert/home_assistant_solarman #697: Solarman inverter chint

Linus-ui, 2024-12-26:
Non ricevo dati dall'inverter

Linus-ui, 2024-12-26:
ho il data logger solarman LSW3 installato su inverter chint CPS SCE5KTL-120,non ricevo dati su home assistant,tutti i valori riportano la dicitura:"sconosciuto".Il fatto che riceva tutte le descrizioni dei valori,mi fa pensare che qualcosa funzioni nelle procedure,ma non vengono caricati i valori,quindi ,presumo,non funziona completamente l'interrogazione del data logger da parte dell'integrazione solarman in home assistant?

davidrapan, 2025-01-26:
Hi @Linus-ui, [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) contains profile dedicated for your inverter.

Linus-ui, 2025-01-26:
Ciao,
grazie per la tua risposta, mi dici anche come si chiama il profilo dedicato all'inverter chint che è contenuto in solarman?
Saluti

davidrapan, 2025-01-26:
`chint_cps-scetl.yaml`

Linus-ui, 2025-01-26:
Dopo aver scaricato solarman, lo configuro, nel menu a tendina che si apre quando si deve inserire la definizione dell'inverter, non trovo il file yaml da te indicato

Linus-ui, 2025-01-26:
Gentilmente, mi puoi dire i passaggi per inserirlo in solarman?

Linus-ui, 2025-01-26:
Nel solarman

Linus-ui, 2025-01-26:
Intendo dire solarman

davidrapan, 2025-01-26:
Did you remove this integration and installed the other one?

Linus-ui, 2025-01-26:
Quella che trovo in hacs, l'ho scaricata pochi minuti fa

Linus-ui, 2025-01-26:
[image]

Linus-ui, 2025-01-26:
Questo è quello che vedo

davidrapan, 2025-01-26:
You need to remove the one installed from HACS and install the other one using first blue button on the page from link i provided you with before: [ha-solarman](https://github.com/davidrapan/ha-solarman)

Linus-ui, 2025-01-27:
Ciao, non ho fatto ancora nulla di quanto da te suggerito, sono fuori al lavoro, vedo questa sera e ti aggiorno

Linus-ui, 2025-01-27:
E ti aggiorno

Linus-ui, 2025-01-27:
Il correttore automatico, fa casini, intendevo dire che poi ti faccio sapere

Linus-ui, 2025-01-27:
Ora manca solo una bella Grafica che visualizzi i dati forniti dall'inverter

davidrapan, 2025-01-27:
Thank you for your kind words @Linus-ui! I'm glad you got it going! 😉

Fabio2411, 2025-04-13:
Ciao David, io ho lo stesso problema ma non riesco a risolverlo, mi da tutto "sconosciuto"

davidrapan, 2025-05-04:
Hi @Fabio2411, in which integration?