StephanJoubert/home_assistant_solarman #703: Entità non numerica occasionale.

lexlutor72, 2025-01-07:
Ho un inverter Zucchetti HYD 6000. La configurazione funziona regolarmente, tutti i sensori vengono riconosciuti. Spesso accade che il dato non viene letto ed il sensore va in modalità "entità non numerica occasionale", per poi ritornare corretto dopo qualche minuto. Allego screenshot. Grazie.
[image]
[image]

freddibn, 2025-01-09:
Ho lo stesso inverter (ZCS HYD6000). 
Tutte le entità sconosciute.
allego lo screen del registro
[image]

Karsto-sys, 2025-01-13:
from what i could understand i have the same problem

davidrapan, 2025-01-26:
Hi guys, I'd like to suggest you to move over to [ha-solarman](https://github.com/davidrapan/ha-solarman)