StephanJoubert/home_assistant_solarman #705: WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml',)

asbor, 2025-02-04:
I was hoping i could get some advice.... im trying to get my solarman integration operational, but im only getting Unknown values...
Inverter specs: Three-phase hybrid inverter Deye SUN-10K 10000W, 2x MPPT
 
Im completely new to home assistant, but as of now, i have managed to install HACS and i have installed the Solarman Integration. Iv configured the logger device serial number, ip address, port and im using the deye_sg04lp3.yaml lookup as per the user manual. I have also tried the all other DEYE lookup files, restarted HA in between each file, and values only change from unknown and unavailable.

Current Setup:
Unraid server
Spaceinvader one HomeAssistant in a box
Solarman integration 1.5.1 installed through HACS

additional spacs on the inverter can be found on https://ecoprodukt.sk/p/87591-trojfazovy-hybridny-menic-deye-sun-10k-10000w-2x-mppt-38993

Let me know if i can help somehow. Thanks. 

**log:** 

2025-02-04 19:18:16.621 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-04 19:18:16.623 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-04 19:18:23.995 INFO (MainThread) [Jellyfin] ---[ START JELLYFINCLIENT ]---
2025-02-04 19:18:24.322 DEBUG (MainThread) [custom_components.solarman] __init__.py:async_setup_entry({'created_at': '2025-02-04T18:17:19.631351+00:00', 'data': {'inverter_host': '192.168.1.219', 'inverter_mb_slaveid': 1, 'inverter_port': 8899, 'inverter_serial': 2110230761, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'Solarman'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JK92WNGF8D3Y86VMTGPJ5T78', 'minor_version': 1, 'modified_at': '2025-02-04T18:17:19.631362+00:00', 'options': {'inverter_host': '192.168.1.219', 'inverter_mb_slaveid': 1, 'inverter_port': 8899, 'inverter_serial': 2110230761, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'Solarman'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': '192.168.1.219', 'unique_id': None, 'version': 1})
2025-02-04 19:18:26.390 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_entry: {'inverter_host': '192.168.1.219', 'inverter_mb_slaveid': 1, 'inverter_port': 8899, 'inverter_serial': 2110230761, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'Solarman'}
2025-02-04 19:18:26.390 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_platform: {'inverter_host': '192.168.1.219', 'inverter_mb_slaveid': 1, 'inverter_port': 8899, 'inverter_serial': 2110230761, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'Solarman'}
2025-02-04 19:18:26.454 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
    handle._run()
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 788, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 551, in async_setup
    await self.__async_setup_with_context(hass, integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2291, in async_forward_entry_setups
    await self._async_forward_entry_setups_locked(entry, platforms)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2302, in _async_forward_entry_setups_locked
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2304, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2383, in _async_forward_entry_setup
    await entry.async_setup(self.hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 551, in async_setup
    await self.__async_setup_with_context(hass, integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 96, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 197, in async_setup_entry
    return await self._platforms[key].async_setup_entry(config_entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 333, in async_setup_entry
    return await self._async_setup_platform(async_create_setup_awaitable)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 363, in _async_setup_platform
    awaitable = create_eager_task(awaitable, loop=hass.loop)
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/config/custom_components/solarman/sensor.py", line 89, in async_setup_entry
    _do_setup_platform(hass, entry.options, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
  File "/config/custom_components/solarman/solarman.py", line 31, in __init__
    with open(self.path + self.lookup_file) as f:

2025-02-04 19:18:26.635 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:_do_setup_platform: async_add_entities
2025-02-04 19:18:26.635 DEBUG (MainThread) [custom_components.solarman.sensor] [<entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unk [... cut ...]
2025-02-04 19:18:56.705 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=15, family=2, type=1, proto=6, laddr=('192.168.1.240', 40020), raddr=('192.168.1.219', 8899)>
2025-02-04 19:18:56.706 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] [2110230761] SENT: a5 17 00 10 45 8c 00 e9 90 c7 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 57 f4 34 3d 15
2025-02-04 19:18:58.399 DEBUG (Thread-6 (_data_receiver)) [custom_components.solarman.solarman] [2110230761] V5_SEQ_NO_MISMATCH: a5 17 00 10 45 03 00 e9 02

davidrapan, 2025-02-05:
Hi @asbor, switch to [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman)

asbor, 2025-02-05:
Hi @davidrapan, That solved the issue ! 

Many thanks !