StephanJoubert/home_assistant_solarman #706: Deye Sun SUN-12K-SG04LP3 - System Work Mode-2 - Time of Use - Modbus Register

okanano, 2025-02-05:
Hi,

The model of my inverter is Deye Sun SUN-12K-SG04LP3, do you know what's the Modbus register address to enable/disable the **"Time of Use"** toggle in the **System Work Mode-2** section?

[image]

davidrapan, 2025-02-05:
Hi @okanano, switch to [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) and you will be able to control your inverter directly using HA entities.

okanano, 2025-02-07:
> Hi [@okanano](https://github.com/okanano), switch to [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) and you will be able to control your inverter directly using HA entities.

Thanks mate! @davidrapan

closed 2025-02-07 by okanano