StephanJoubert/home_assistant_solarman #711: Unable to connect using local dns name instead of ip

LuisDiazUgena, 2025-02-21:
Hi!

Just tried to setup the integration using a local dns name instead of an IP and the integration fails to connect.

It works with the IP:

[image]

but does not with the local dns name:

[image]

HA config has the local dns as the dns server:

[image]

And the name resolves in the home network:

[image]

It's not a huge deal but just posting for the devs to know.

davidrapan, 2025-03-11:
Hi @LuisDiazUgena, you can try [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman)

closed 2025-03-25 by LuisDiazUgena

davidrapan, 2025-04-03:
If you are tech-savvy, [here](https://github.com/davidrapan/ha-solarman/discussions/183) is how to merge your previous data.