StephanJoubert/home_assistant_solarman #713: complet Deye Cloud-Free opreation, any change?

coolje, 2025-02-23:
I' using this integration quite a long time in homeassist successfully read out a 300W (single tracker) and an 800W (dual) deye microinverter. Thank the makers your for this.
But for operation of the inverter i still need the deye-cloud for the correct setting of the timeserver and the dayly reset of the produces Power. 
(if not both is not resettet)

Is there any change that a "dummy-cloud" could be integrated to your package like in the github project.
https://github.com/Hypfer/deye-microinverter-cloud-free
To operate the deye converter completetly cloudfree in homeassist. 
Alone the setting of an alternative local timeserver could help a lot.
(unfortunately the above "cloud-free" githup is to complicated for me to integrate it in homeassist for my own)

davidrapan, 2025-03-11:
Hi @coolje, install [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) and follow example in https://github.com/davidrapan/ha-solarman/issues/348

coolje, 2025-03-13:
This alternative you pointet me to,  is super. Thank you very much.