StephanJoubert/home_assistant_solarman #715: Support for monitoring individual battery modules in Deye Hybrid 5kW

adambrejniak, 2025-02-25:
Hello,

First of all, thank you for your work on this integration!

I have been using the integration for several months now, and it has been working well. I am connecting to my Deye Hybrid 5kW inverter using a USB to RS485 converter.

I have three Deye batteries, and currently, I can monitor overall battery performance. However, I would like to see detailed data for each individual battery module. Specifically, I would like to track:

Voltage per battery module

Current per battery module

State of charge (SOC) per battery module

Temperature per battery module

Would it be possible to add support for monitoring individual battery modules in the integration? If there is any way I can assist with testing or providing data, please let me know.

Thanks in advance!

Best regards

davidrapan, 2025-03-11:
Hi @adambrejniak, you can try [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman)

davidrapan, 2025-10-17:
Hi @adambrejniak, it's supported, but only for three-phase inverters (registers 10000+), atm.

sigurdi, 2025-10-26:
Check out this respository for using esp32 and esphome to read internal CAN between master pack and slave packs for information you are looking for, indevidual voltage.... 
I see in the yaml that it is setup for 2 packs but you can add for the rest since address is set, 
can_id: 0x02098001  = pack 1
can_id: 0x02098002 = pack 2
can_id: 0x02098003 = pack 3....... 
Internal CAN uses CAN FD and external the older CAN protocoll. 

Home assistant and Homee is compatible with ESPHome

That is why set setup with ESP32 is with two CAN chip where the external supports CAN FD and interal supports regualr CAN. 
https://github.com/Psynosaur/esphome-deye-bms?tab=readme-ov-file