StephanJoubert/home_assistant_solarman #718: Sofar Hybrid inverter MS3000EP

marichardson535, 2025-02-28:
Getting all sorts of issues where most data is 'unavailable'.

Not sure if im doing something wrong ...

Found this ...

Logger: homeassistant.util.loop
Source: util/loop.py:137
First occurred: 13:41:02 (1 occurrences)
Last logged: 13:41:02

Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/sofar_hyd3k-6k-es.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2377, in _async_forward_entry_setup await entry.async_setup(self.hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 548, in async_setup await self.__async_setup_with_context(hass, integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context result = await component.async_setup_entry(hass, self) File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 96, in async_setup_entry return await hass.data[DATA_COMPONENT].async_setup_entry(entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 193, in async_setup_entry return await self._platforms[key].async_setup_entry(config_entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 333, in async_setup_entry return await self._async_setup_platform(async_create_setup_awaitable) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 363, in _async_setup_platform awaitable = create_eager_task(awaitable, loop=hass.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/solarman/sensor.py", line 89, in async_setup_entry _do_setup_platform(hass, entry.options, async_add_entities) File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file) File "/config/custom_components/solarman/solarman.py", line 31, in __init__ with open(self.path + self.lookup_file) as f:

srk23, 2025-03-06:
Getting exactly the same errors recently - was working previously.  I think related to https://github.com/StephanJoubert/home_assistant_solarman/issues/717

grimuk06, 2025-03-29:
im getting the same issue with my HYD 3600 ES, although I've only just installed it, so I may have made a mistake in set up. 
I've upgraded to the latest firmware 4.10, and it hasn't made any difference.

grimuk06, 2025-03-29:
try this HACS [https://github.com/davidrapan/ha-solarman?tab=readme-ov-file]