StephanJoubert/home_assistant_solarman #723: No connection to my Sofar ESI-5K-S1 inverter via the logger with S/N 27xxxxxxxx (10 digits)

uswong, 2025-03-07:
Hi, installation and configuration went very smoothly and it seems to connect successfully.  All the sensors' values will unknown and after a few minutes, 'Sofar status_connection' shown as disconnected.  I have double checked the logger ip address and s/n and found to be all correct.  The SM 27xxxxxxxx of my logger seem to be not supported out the box as explained in the README and advised to use protocol V4, but how do I do that?

[image]

MrGuiseppe, 2025-03-07:
I have same problem with S/N 17xxxxxxxx

uswong, 2025-03-07:
Just want to add that the firmware used by my logger with S/N 27xxxxxxxx is LSW3_15_MQTT_270A_1.10

Thales79, 2025-03-10:
Mám stejný problém.

uswong, 2025-03-11:
Have you try using `sofar_g3hyd.yaml` profile?  If it still does not work, you could try improved version of Solarman integration done by David Rapan, https://github.com/davidrapan/ha-solarman. 

I got it working with David's version and using `sofar_g3hyd.yaml` profile.

All the best.

MichaelPloese, 2025-04-14:
Same issue happened to me after the loggers firmware was updated to last version. Log shows an python error.

davidrapan, 2025-05-05:
Hi @MichaelPloese, some of the newest firmware version have port 8899 closed. You need to look for a way how to open back again in the settings or ask solarman for firmware downgrade.

kromaaa, 2025-10-25:
@tonytvo168 no, it stays as is. You are not allowed to change ip of the server - this field is disabled

schmttc, 2026-10-09:
> 4\. Have you try using `sofar_g3hyd.yaml` profile?  If it still does not work, you could try improved version of Solarman integration done by David Rapan, https://github.com/davidrapan/ha-solarman.
>         I got it working with David's version and using `sofar_g3hyd.yaml` profile.
>         All the best.

This worked for me with a SOFAR ESI 6K-S1 invertor and logger LSW3_15_MQTT_270A_1.22 SN:31*.

Thank you @uswong :)