StephanJoubert/home_assistant_solarman #730: Second battery not showing on Sofar HYD15-KTL-3PH

Gibol, 2025-04-02:
https://github.com/StephanJoubert/home_assistant_solarman/commit/e3e5b11376565bddf54aa41b6461588d47954265
Says: Remove battery 2 sensors (they weren't actually being read from the registers)

That was indeed the case with 1.0.0 version I had previously, battery 2 was uknown.
Why this values are not being read?
I can see them in solarman app just fine.

What does the "Number of battery packs" slider does?
Should I expect second battery to show up when I set this slider to 2?

gkasprow, 2025-05-01:
I added second battery registers to yaml files but the status is unknown...

      - name: "Battery2 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 2
        registers: [ 0x060D ]
        icon: 'mdi:battery-charging'

      - name: "Battery2 SOC"
        class: "battery"
        state_class: "measurement"
        uom: "%"
        scale: 1
        rule: 1
        registers: [ 0x060F ]
        icon: 'mdi:battery'

      - name: "Battery2 SOH"
        class: "battery"
        state_class: "measurement"
        uom: "%"
        scale: 1
        rule: 1
        registers: [ 0x0610 ]
        icon: 'mdi:battery'

      - name: "Battery2 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.01
        rule: 2
        registers: [ 0x060C ]
        icon: 'mdi:current-dc'

      - name: "Battery2 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [ 0x060B ]
        icon: 'mdi:battery'

      - name: "Battery2 Temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x0607 ]
        icon: 'mdi:battery'
        
      - name: "Battery2 Number of Cycles"
        class: ""
        state_class: "measurement"
        uom: "cycle"
        scale: 1
        rule: 1
        registers: [ 0x0611 ]
        icon: 'mdi:battery'

gkasprow, 2025-05-01:
Of course, I also modified the address range
  - start: 0x0604
    end: 0x0611 # end of second battery after this continue battery pack 2,3,4

gkasprow, 2025-05-01:
Oh, it works
I wasn't patient enough :)

[image]