StephanJoubert/home_assistant_solarman #733: Hybrid 6K  SG03LP1 EU  + micro inverter

Wingman2, 2025-05-04:
Hi 
I have a SUN-6K-SG03LP1-EU

It was working very good with this integration until I added 3 micro inverters and connected the to the GEN port.

The total energy production is not adding the micro inverter productions like the official website does.

One item I can see is that micro inverter production is being subtracted from house consumption. Totals keep drifting apart. I can´t find values that are similar to the website. 

Battery Charge and Discharge look ok
Same for Import and export.

I don´t seem to find an accumulation for the micro inverters production

[image]

[image]

davidrapan, 2025-05-05:
Hi @Wingman2, if you want you can try [ha-solarman](https://github.com/davidrapan/ha-solarman) and see if **Total Generation** has any meaningful value for you. I am myself very curious because in 3 phase version is generator port already accounted in **Total Production**, etc.

Wingman2, 2025-05-06:
Hi @davidrapan 

Thanks for the recommendation. I like ha-solarman

I had to wait to have some data to compare

Looks like Solar Production is going into 2 "buckets".   Total Production and Total Generation. 

Generation is where my micro inverter production is adding up. And Total Production is just showing the mppts. Knowing that I can create a new variable to keep track of the total. It's not a grate solution but at least numbers have some meaning now.

davidrapan, 2025-05-13:
Great! Thanks for letting me know. Power losses looks somewhat alright but the other two are messed up as expected because their calculation is missing that energy from microinverter.