StephanJoubert/home_assistant_solarman #736: Unused yaml configuration platform solarman in Home Assistant 2025.5.1

ArmenVanlasso, 2025-05-10:
`It's not possible to configure solarman sensor by adding platform: solarman to the sensor configuration. Please check the documentation for more information on how to set up this integration.

To resolve this:

Remove platform: solarman occurrences from the sensor: configuration in your YAML configuration file.
Restart Home Assistant.
Example that should be removed:

sensor:
  - platform: solarman

[image]

`

davidrapan, 2025-05-10:
Hi @ArmenVanlasso, integration is not configurable from `configuration.yaml`

ArmenVanlasso, 2025-05-10:
@davidrapan hah?
Is in documentation:

[image]

davidrapan, 2025-05-10:
Sure it can be there but the code doesn't support it anymore.

davidrapan, 2025-05-10:
I don't know as I'm not using it and I don't think there is any other way than changing it in the code itself. Also this integration is not very performant so lowering the update interval is not good idea. But you can try [ha-solarman](https://github.com/davidrapan/ha-solarman) which has 5s update interval.