StephanJoubert/home_assistant_solarman #757: MAJ HA 2025.8.0

bgt25870, 2025-08-08:
Bonjour, lorsque je met a jour mon ha en version 2025.8.0, je perd la connexion avec mes onduleurs qui sont connecté avec integration Solarman, problème de connexion qui n'aboutit pas, si je reviens a la version 2025.7.4 pus de problème sans rien changer, quelqu'un a til rencontré le meme problème  ?

CrazyUs3r, 2025-08-16:
Hi @bgt25870 , switch to [https://github.com/davidrapan/ha-solarman](url)

bgt25870, 2025-08-19:
Hello, when I update my HA to version 2025.8.2, I lose the connection with my inverters which are connected with Solarman integration, connection problem which does not succeed, if I return to version 2025.7.4 no more problem without changing anything, has anyone encountered the same problem?

acobo, 2025-08-19:
well, actually not, I thougth it was 2025.8.2 becouse is very recent, but I checked it and it is actually 2025.8.0. In any case, this integration did not work, and https://github.com/davidrapan/ha-solarman  did work in 2025.8.0.